HViktorTsoi/PV-LIO #1: robosense timestamp

chengwei0427, 2023-07-15:
Hi, @HViktorTsoi 
Thanks for your nice work!
I'm confused about `the robosense data` you shared, whether the header timestamp is frame head or frame end?

HViktorTsoi, 2023-07-15:
Hi, the raw data of robosense is collected from an old rs16 LiDAR with earlier version of the LiDAR driver, and the timestamps of the points in 1 frame may not be stored in order in some circumstances, therefore we write a [converter](https://github.com/HViktorTsoi/PV-LIO/blob/master/src/converter/rs_to_velodyne.cpp) to handle this. In the converted point cloud msg,  **the msg header timestamp is the time of the earliest point in 1 frame**.

HViktorTsoi, 2023-07-15:
The converter basically conver the robosense point cloud to a standard velodyne point cloud format, you may refer to this repo https://github.com/HViktorTsoi/rs_to_velodyne

HViktorTsoi, 2023-07-15:
If you are testing the shared data with your own algorithm, you may run the converer node alone, and then use the converted point cloud msg and the raw imu msg for testing.

chengwei0427, 2023-07-15:
Hi, Dear author,
The frame header timestamp is frame end, and I can run my program with the shared data successful.
[image]
Problem solved!
Thanks again.

closed 2023-07-15 by chengwei0427