HViktorTsoi/PV-LIO #2: Is sensors' hard synchronization necessary?

msusing, 2023-07-15:
Thank you for your work. 
I'm not familiar with the fast-lio series. I have a velodyne lidar and an independent IMU. But they are not hardware-wise synchronized. Is that OK to run the code?  
Hope for your reply. Thank you very much.

HViktorTsoi, 2023-07-15:
Hi, it would be best (and we strongly recommend) if you could provide hard synchronization. If that's really not possible, you could try setting `time_sync_en` to `True` in the config yaml files. While the algorithm would still run, its performance under fast motion or other extreme conditions cannot be guaranteed.

You may refer to the [issues in FAST-LIO](https://github.com/hku-mars/FAST_LIO/issues), many of which are about hardware synchronization.

msusing, 2023-07-15:
Thank you for your reply. Doe it mean after turn on `time_sync_en` I need to specify a time offset?

HViktorTsoi, 2023-07-15:
No, with `time_sync_en` set to true, it will automaticly estimate the approximate offset between LiDAR and IMU

We keep the `time_offset` parameter in the code for debugging, if you are sure that the time offset between LiDAR and IMU remains constant, you may set `time_sync_en` to `false` and then set the known `time_offset` parameter from your setup.

Freddd13, 2023-07-15:
Got it. Thank you very much.

closed 2023-07-15 by msusing