HViktorTsoi/PV-LIO #3: Robosense data failed

CharlieV5, 2023-07-15:
通过百度网盘下载的三个Robosense数据
[image]
启动mapping_robosense.launch后崩溃：
[image]
[image]

CharlieV5, 2023-07-15:
系统为Ubuntu20.04, Noetic

Yeah2333, 2023-07-15:
我也遇到了一样的问题。
[image]

HViktorTsoi, 2023-07-15:
We tested our implement with Ubuntu18.04 and row melodic, this may be some issue with the default Eigen version, we will soon test it on 20.04

HViktorTsoi, 2023-07-15:
> 系统为Ubuntu20.04, Noetic

Could you tell us which version of Eigen you are using?

CharlieV5, 2023-07-15:
> > 系统为Ubuntu20.04, Noetic
> 
> Could you tell us which version of Eigen you are using?

Eigen 3.4.0

JJho1314, 2023-07-15:
Hi，I meet same trouble with ubuntu 18.04 and ROS melodic, and the version of my Eigen is 3.3.7, thanks

HViktorTsoi, 2023-07-16:
Hey guys, I've tested with Eigen 3.3.7 / 3.3.4 and it works fine with both version. What is your PCL version? And could you upload the whole output from the first few seconds of the algorithm's execution?

Yeah2333, 2023-07-16:
I meet same trouble with ubuntu18.04, ros melodic, eigen 3.3.7

Yeah2333, 2023-07-16:
The code output No effective points , and then crush. I check the code, it seems no enough measurement ptpl. And i check the code , in this line https://github.com/HViktorTsoi/PV-LIO/blob/a4d2dbf0f5a65446cd36af3dfca7416d9ea38919/include/voxel_map_util.hpp#L703C36-L703C36. This line check p2plane dist, but plane.radius get a very low value.
[image]
So, i edit this line just using 0.1 instead of plane.radius. But the code also crush.
And i found rviz have some wrong plane. So i think code may have some troubel in plane extract. 
[image]

Yeah2333, 2023-07-16:
By the way ,before code crush. i have some eigen aligned problem. I fix this using align allocater

885288421, 2023-07-17:
Hi，This is an excellent project, thank you for your contribution！but I meet same trouble with ubuntu 20.04 and ROS nortic, and the version of my Eigen is 3.3.7, pcl is 1.10.0,I record my rosbag outdoors,I would like to know how to fix this.thanks

HViktorTsoi, 2023-07-17:
Hi, a tiny but important question, for those using the demo rosbag dataset, did you play the bags in order with `rosbag play *.bag`? There may be some issues if you play the second or the third bag alone, because the LIO currently requires static initialization.

HViktorTsoi, 2023-07-17:
> Hi，This is an excellent project, thank you for your contribution！but I meet same trouble with ubuntu 20.04 and ROS nortic, and the version of my Eigen is 3.3.7, pcl is 1.10.0,I record my rosbag outdoors,I would like to know how to fix this.thanks

Hi, could you upload the rosbag and LiDAR-IMU extrinsic parameter for us to debug? That will be a lot more efficient for addressing this issue.

HViktorTsoi, 2023-07-17:
> The lidar is robosense,but we used a rospackage rs_to_velodyne to convert the data ,the imu topic is"/sensor/imu/imu_data",the lidat topic is "/sensor/lidar/points",the rosbag url is :https://pan.baidu.com/s/1BmncWoxoeeMgn1BMP7VUnA password: 3nst. extrinsic_T: [-0.376489, -0.0505456, -0.109697 ] extrinsic_R: [ 1, 0, 0, 0, 1, 0, 0, 0, 1 ] I hope to hear good news,thanks!

Hi, we foud the problem, it's because of the parameter setting, the default parameter we provide is for narrow indoor environment, but your test scene is a largescale outdoor scene(and the LiDAR is sparse), there for the `down_sample_size` and `voxel_size` are too small for such scenario. After increasing these two parameter, it runs normally.
[image]
[image]

We have provide a launch and corresponding config file called `robosense_outdoor.yaml`, please pull the update repo, and modify the config file path in `robosense.launch` to   
```
    <rosparam command="load" file="$(find pv_lio)/config/robosense_outdoor.yaml"/>
```
You may also try to tune the parameter setting for better performances.

885288421, 2023-07-17:
hi,thanks for your help. the problem still exists.Do you test on noetic,I will change the environment to test the rosbag .

HViktorTsoi, 2023-07-17:
> hi,thanks for your help. the problem still exists.Do you test on noetic,I will change the environment to test the rosbag .

We tested with melodic.

CharlieV5, 2023-07-18:
I have test in Ubuntu18.04, ROS melodic. The program runs well. So probably the problem is about the system environment.

HViktorTsoi, 2023-07-18:
Hey guys, we reproduce this bug on Ubuntu 20.04, it is because some numerical type convention issue of GCC 9.3 and the default Eigen of Ubuntu 20.04, which causing abnormal result while calculating the voxel coordinates of points. 

We have pushed a fixed for this issue. Please pull the latest code and test it. It would be very nice if you let us know the test result.

closed 2023-07-18 by CharlieV5

alwynmathew, 2023-09-29:
@HViktorTsoi I get the below error with `roslaunch pv_lio mapping_robosense.launch` and `rosbag play staircase_crazy_rotation/2022-08-30-20-33-52_0.bag`.

Ubuntu 20.04
ROS Noetic
Eigen 3.3.7
PCL 1.10

```
pv_lio_node: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, 
MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 36; int MatrixOrArrayOptions = 0]: 
Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && 
"this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " 
**** READ THIS WEB PAGE !!! ****"' failed.
[pv_lio_node-1] process has died [pid 23511, exit code -6, cmd /home/alwyn/catkin_pv/devel/lib/pv_lio/pv_lio_node 
__name:=pv_lio_node __log:=/home/alwyn/.ros/log/95c3ca40-5e20-11ee-9ce3-2b06debad6a1/pv_lio_node-1.log].
```

alexjunholee, 2024-05-20:
@alwynmathew try upgrading to C++17, it is explained in that eigen website link.