HViktorTsoi/PV-LIO #5: run error。

linClubs, 2023-07-18:
我在ubuntu20.04，跑自己的voledyne数据，终端输入如下：我应该怎么处理这种情况

[image]

linClubs, 2023-07-18:
这是我运行环境 ubuntu-20.04， eigen-3.3.7， pcl-1.10， gstam-4.1.1，使用的velodyne雷达，我应该怎么解决上面的问题，谢谢

[image]

HViktorTsoi, 2023-07-18:
Did it fail on the demo robosense rosbag?

linClubs, 2023-07-18:
感谢您的回复。我测试了2022-08-30-20-37-43_2.bag数据包
报错一样的， 然后我增大了 down_sample_size与 voxel_size，还是一样的错误。rviz也没显示
[image]

增大了 down_sample_size与 voxel_size，我测试我自己的velodyne数据集，rviz有显示点云，但终端也报同样的错误了
[image]

linClubs, 2023-07-18:
> Did it fail on the demo robosense rosbag?
感谢您开源的工作，我刚刚测试一下2022-08-30-20-37-43_2.bag数据包。这是终端输出结果https://github.com/HViktorTsoi/PV-LIO/issues/5#issuecomment-1639282623，还是报错

HViktorTsoi, 2023-07-18:
> 感谢您的回复。我测试了2022-08-30-20-37-43_2.bag数据包 报错一样的， 然后我增大了 down_sample_size与 voxel_size，还是一样的错误。rviz也没显示 [image]
> 
> 增大了 down_sample_size与 voxel_size，我测试我自己的velodyne数据集，rviz有显示点云，但终端也报同样的错误了 [image]

Please test the 2022-08-30-20-33-52_0.bag, 2022-08-30-20-37-43_2.bag can't be played alone.

HViktorTsoi, 2023-07-18:
Seems like a common issue for Ubuntu 20.04, we are trying reproducing the issue in this environment.

linClubs, 2023-07-18:
> Seems like a common issue for Ubuntu 20.04, we are trying reproducing the issue in this environment.

我测试了2022-08-30-20-33-52_0.bag。跟测试我的数据一样，跑几秒终端就报错了，可能在20.04上的还需要调试一下，感谢您的开源
[image]

linClubs, 2023-07-19:
> We've pushed a fix for this, please refer to [#3 (comment)](https://github.com/HViktorTsoi/PV-LIO/issues/3#issuecomment-1640066605)
感谢大佬的工作。已经测试通过^_^
[image]

closed 2023-07-21 by HViktorTsoi