HViktorTsoi/PV-LIO #7: Any suggestion on memory usage for large-scale dataset?

wlt027, 2023-07-20:
Hi,thanks for nice work!
I have test my own data and get good result.But for large-scale data or long-term data,I found that this process use too much memory,so could you please give some suggestions? 
[image]

HViktorTsoi, 2023-07-20:
Hello, excessive memory consumption is caused by the large size of the VoxelMap. The structure of the VoxelMap is actually an std::unordered_map. You can try keeping track of the occupancy status of each map cell and remove those cells that have been unoccupied for a long time(or too far away), thus maintaining a smaller local map. We are also planning to add this feature.

HViktorTsoi, 2023-08-02:
Besides, if dense point cloud publishing is enabled, the RVIZ process may also consume a lot of memory.