HViktorTsoi/PV-LIO #10: which kind of IMU do you recommend?

AcamelQ, 2023-08-10:
Thanks for your excellent work.I am interested in your system hardware requirement. which kind of IMU did you employ to acquire the Robosense 16 Rosbag(multi-floor)？as my imu usually will have large drift when have large rotation.it is very helpful to have your answer.

AcamelQ, 2023-08-14:
Thanks for your help. which kind of algorithm you used for the calibration of extrinsic R/T matrix for lidar and IMU? and also for the noise of IMU? when i tested on my multi-floor dataset, there was a huge drift when my robots was climbing the stairs.

HViktorTsoi, 2023-08-20:
Hi, sorry for the late reply. We did not do offline LiDAR-IMU calibration or IMU noise calibration in our sensor setup. We just manually measure the LiDAR-IMU extrinsic  from CAD as the initial value, and then enable the [online extrinsic refinement option](https://github.com/HViktorTsoi/PV-LIO/blob/494c27f62f7333d6cc32c26dccaa8d8a971c2c2d/config/robosense.yaml#L25) in the config files.

But please make sure that your initial extrinsic value is mostly correct, especially the rotation part.

HViktorTsoi, 2023-08-20:
Also the hardware time synchronization between LiDAR and IMU is important.

AcamelQ, 2023-08-21:
thanks for your help.

85256638, 2024-11-17:
> Also the hardware time synchronization between LiDAR and IMU is important.

@HViktorTsoi 
Hi, if there's time delay between the IMU and Lidar, where do I input this delay? For Fast-LIO2, the config.yaml file has     `time_offset_lidar_to_imu`