HViktorTsoi/PV-LIO #12: Reducing Drift with High Compute

satyajitghana, 2023-09-01:
Hi,

I am trying to use PV-LIO on a scan with a lot of very small rooms. The thing is i don't want realtime mapping, but i want the drift to be as less as possible. I can run the algorithm on 96 cores / 192 cores with 1024GB of RAM. What are the parameters that i need to tune so it makes use of all the compute available?

here's what i tried

- increase openmp number of cores to max: this makes sure to use all cores, but it only uses like 50% of each core.
- decrease the voxel size: i tried 0.1 and 0.05 and downsample size to 0.05, but still there is slight drift overall.
- increase max iterations: i tried increasing this to 100 and then to 500, but there wasn't any change from this, maybe slightly but not major
- increase number of voxel layers: this absolutely did nothing
- decrease the planar threshold: if i try to reduce this below 0.05 there is drift

So what parameters should i try to tune?

I have already tried things like SC (Scan Context) and LC (Loop Closure) algorithms (https://github.com/gisbi-kim/FAST_LIO_SLAM) that were suited for FASTLIO2 but they work well with PV-LIO https://github.com/hku-mars/Point-LIO as well. but they also are not able to fix the drift.

Also I've tried HBA (Hierarchical Bundle Adjustment) (https://github.com/hku-mars/HBA) , it does solve the drift, but it adds too much noise to the point cloud to a point that the point cloud is unusable.

Also note. I am using MID360 with inbuilt imu, should i try out an external IMU? will that help?

Some areas its really good

[image]

some areas its really bad and drifts, its mostly when we go from a large room to a small room or vice versa

[image]

[image]

also NOTE: FASTLIO2 does not drift in the above case, although it drifts as soon as we enter a small room in some cases, but the above case FASTLIO2 is able to solve.

- I was also thinking if it would be possible to do something like what Point-LIO does? increasing the rate of IMU might solve to some extent?
- also maybe STD might help? https://github.com/hku-mars/STD. I've seen an implementation of STD with FASTLIO2, but its just shit.

aditdoshi333, 2023-09-19:
Facing similar issue. Waiting for the author's reply

Lelehel, 2023-12-01:
Maybe it's just MID360's poor IMU?

Can you share mid360 config .yaml?

satyajitghana, 2024-01-22:
@Lelehel using the one from : https://github.com/hku-mars/FAST_LIO/blob/main/config/mid360.yaml