HViktorTsoi/PV-LIO #13: nMeasurements < nDof;

aditdoshi333, 2023-09-14:
Facing this error a lot of times. Can you please explain why it is happening and what scene it can happen?

Please share some light to solve this issue.

CharlieV5, 2023-09-16:
I have encountered this issue too, with the rotating velodyne LiDAR. This often happens in narrow space, maybe . Looking forward to the author's explaining.

satyajitghana, 2023-09-19:
The author has removed code for `if (nMeasurements < nDof)` which was there in FASTLIO2.

CharlieV5, 2023-09-19:
> The author has removed code for `if (nMeasurements < nDof)` which was there in FASTLIO2. due to integration with `VoxelMap`.

Could you tell us where the code for if (nMeasurements < nDof) in Fast-LIO2 can be found? I tried but not see it in the latest version.

satyajitghana, 2023-09-20:
You can take a look at this: https://github.com/HViktorTsoi/PV-LIO/blob/494c27f62f7333d6cc32c26dccaa8d8a971c2c2d/include/IKFoM_toolkit/esekfom/esekfom.hpp#L1756

Actual Code: https://github.com/hku-mars/FAST_LIO/blob/478d436813b966df94fb91dd759f477f1a246296/include/IKFoM_toolkit/esekfom/esekfom.hpp#L890

You can see lot of lines commented out in PV-LIO which were actual parts of FASTLIO2.

Also another thing to note is that, you can actually disable the voxel based mapping in PV-LIO, which will then rely on the IKFoM based method only using IMU, and if you do that it WILL DRIFT! which it shouldn't, given that FASTLIO2 does not drift.