HViktorTsoi/PV-LIO #14: PCD Output - pcd_save_en: true

Snowbound9173, 2023-09-24:
Hi! Great job with your project! I've only just tried it and it looks so good. 

Here's a screenshot i took using a Livox HAP and you really can see it tracking the walk. 
[image]
And another from top down: 
[image]

The configuration that includes quite a bit from FAST-LIO2 mentions PCD save but for the life of me i can't seem to find if this would actually do anything. I'd assume that due to the updated pipeline that may not have been factored in? But I'm not really sure. 
[image]

Is there any way to output a PCD?

Again thanks for publishing such a great quality project.

redheli, 2023-10-08:
can share the yaml file for hap? Thanks

Snowbound9173, 2023-10-08:
Hi, 

Sure, but it's exactly the same as the Avia

```
common:
    lid_topic:  "/livox/lidar"
    imu_topic:  "/livox/imu"
    time_sync_en: false         # ONLY turn on when external time synchronization is really not possible

preprocess:
    lidar_type: 1                # 1 for Livox serials LiDAR, 2 for Velodyne LiDAR, 3 for ouster LiDAR,
    scan_line: 6
    scan_rate: 10
    blind: 0.5
    point_filter_num: 1

mapping:
    down_sample_size: 0.4
    max_iteration: 4
    voxel_size: 2.0
    max_layer: 4                # 4 layer, 0, 1, 2, 3
    layer_point_size: [ 5, 5, 5, 5, 5 ]
    plannar_threshold: 0.01
    max_points_size: 1000
    max_cov_points_size: 1000

    fov_degree:    360
    det_range:     100.0
    extrinsic_est_en:  false      # true: enable the online estimation of IMU-LiDAR extrinsic,
    extrinsic_T: [ 0.04165, 0.02326, -0.0284 ]
    extrinsic_R: [ 1, 0, 0,
                   0, 1, 0,
                   0, 0, 1]

noise_model:
    ranging_cov: 0.02
    angle_cov: 0.05
    acc_cov: 0.5
    gyr_cov: 0.25
    b_acc_cov: 0.0043
    b_gyr_cov: 0.000266

publish:
    pub_voxel_map: false
    publish_max_voxel_layer: 2         # only publish 0,1,2 layer's plane
    path_en:  true
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

I guess I was just asking, is there an inbuilt way to save the result to a PCD for later viewing?

redheli, 2023-10-09:
@Snowbound9173 to save point cloud to pcd, check my PR https://github.com/HViktorTsoi/PV-LIO/pull/15