HViktorTsoi/PV-LIO #17: Questions about the process of cov transmission at Lidar point.

hr2894235132, 2023-11-06:
您好，感谢您很好的工作！我在这里有个不懂的地方：关于`transformLiDARCovToWorld`函数中，lidar to body的cov传播过程中，我注意到你在公式中*的是以下两个量。括号里是英文翻译，原谅我英文不是很好。（Hello and thank you for a great job! I don't understand here: regarding the cov propagation of lidar to body in the 'transformLiDARCovToWorld' function, I noticed that you have the following two quantities in the formula.）
```
M3D il_rot_var = kf.get_P().block<3, 3>(6, 6);
M3D il_t_var = kf.get_P().block<3, 3>(9, 9);
```
这是我在之前自己添加imu的时候没有注意到的地方，想问一下对于voxelmap中的state(StatesGroup)中18维的cov，好像并没有定义的外参的var，我该从哪获得这个信息呢，或者我该参考一些什么公式呢？希望您可以在百忙中抽空回复一下，万分感谢！（This is something I didn't notice when I added imu before, I want to ask where do I get this information for the 18-dimensional `cov` in the `state(StatesGroup)` in voxelmap, there seems to be no defined external parameter var, or what formula should I refer to? I hope you can take time out of your busy schedule to reply, thank you very much!）

AcamelQ, 2023-11-28:
Hello, i also have been investigating this issue. this should be the key difference between original voxel_map and pv_lio. i am wondering that this the reason why the author choose ikfoM as the filter rather than employing ieskf(in plain code). Lol, hope you have some advancement about this issue.(actually, the authors are from BH, so chinese is definately okay here). if you are good with it, i would like to exchange contact infor with you. sincerely regards.

hr2894235132, 2023-11-29:
> Hello, i also have been investigating this issue. this should be the key difference between original voxel_map and pv_lio. i am wondering that this the reason why the author choose ikfoM as the filter rather than employing ieskf(in plain code). Lol, hope you have some advancement about this issue.(actually, the authors are from BH, so chinese is definitely okay here). if you are good with it, i would like to exchange contact infor with you. sincerely regards.

Great, my email address is 2894235132@qq.com, and it's great to have someone to talk to and discuss. Can I contact you by email? Or the other way around.（我的邮箱是2894235132@qq.com，有人一起交流讨论太好了。能否通过邮箱联系您呢？或者其他方式都可以。）

hr2894235132, 2023-11-29:
> it will be more convenient to chat with QQ , Lol. so may i follow your QQ number

okok，my qq:2894235132